ROBOTIS-GIT/turtlebot3_machine_learning #1: Error: roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch

harry21105, 2018-07-30:
Hi :)
I got an error while executing the command 
`roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch`

Error: 
```
IOError: [Errno 21] Is a directory: '/home/cow/catkin_ws/src/turtlebot3_machine_learning-develop/turtlebot3_dqn/src/turtlebot3_dqn'
[turtlebot3_dqn_stage_1-1] process has died [pid 14409, exit code 1, cmd /home/cow/catkin_ws/src/turtlebot3_machine_learning-develop/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_1 __name:=turtlebot3_dqn_stage_1 __log:=/home/cow/.ros/log/bc24a084-93e0-11e8-b6dd-34e6addf1aec/turtlebot3_dqn_stage_1-1.log].
log file: /home/cow/.ros/log/bc24a084-93e0-11e8-b6dd-34e6addf1aec/turtlebot3_dqn_stage_1-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
I have followed the installation in the e-manual.
Also cloned the package into catkin_ws/src ,then executed catkin_make.
Look forward to your reply ;)

kijongGil, 2018-07-30:
Hi @harry21105 :)
Now your package folder name is 'turtlebot3_machine_learning-develop'.
Change the this name you cloned to 'turtlebot3_machine_learning'.

Thanks,
Gilbert.

harry21105, 2018-07-30:
Sorry does not understand :(
You mean rename the folder 'turtlebot3_machine_learning-develop'
 to 'turtlebot3_machine_learning' ?

harry21105, 2018-07-30:
hi, problem is solved :)
Thank you!

closed 2018-07-30 by harry21105