ROBOTIS-GIT/turtlebot3_machine_learning #2: ValueError: Error when checking: expected to have shape but got array with shape

shyeonaaa, 2018-08-04:
Hi! Thanks for uploading the good code. Thanks to this, I'm studying reinforcement learning on ROS.
But there was a problem running the program.

[   roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch   ]
If this code is executed, the following error occurs after 1 learning.

ValueError: Error when checking : expected dense_1_input to have shape (None, 26) but got array with shape (1, 362)
[turtlebot3_dqn_stage_1-1] process has died [pid 32123, exit code 1, cmd /home/hyeonaaa/ardrone_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_1 __name:=turtlebot3_dqn_stage_1 __log:=/home/hyeonaaa/.ros/log/f2b4bfb8-97d1-11e8-af19-6045cb6a3a54/turtlebot3_dqn_stage_1-1.log].
log file: /home/hyeonaaa/.ros/log/f2b4bfb8-97d1-11e8-af19-6045cb6a3a54/turtlebot3_dqn_stage_1-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

If you know the cause of the error and the solution, 
I'd appreciate it if you let me know.

I use ROS kinetic, Ubuntu 16.04, cuda 8.0, cudnn 6.0, tensorflow-GPU 1.4.0
not using anaconda packages.

harry21105, 2018-08-05:
hi :)
I had this problem too ,but I solved it somehow.
Change ` state_size = 26` to  `state_size = 326` at line 151.
Hope it worked for you.

shyeonaaa, 2018-08-06:
Oh thanx
I Changed state_size=26 to 362 at line 151. 

Also, This has happed.
[ImportError: `save_model` requires h5py.]

This has solved the problem.
pip install --upgrade h5py

kijongGil, 2018-08-06:
Hi shyeonaaa :)
The meaning of `stats_size = 26` is what input data to use. 
I used 24 LDS values(0, 15, 30 ... 345 degree), distance to goal and angel to goal.
Turtlebot3's LDS default is set to 360.
You can change LDS sample 24 at this line
https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro#L111
Also, you want to see the visualization of LDS, change the this parameter to `true` at https://github.com/ROBOTIS-GIT/turtlebot3/blob/467c76bc4fa2e34162f57107388839d82d3bcc0e/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro#L3

Thanks, 
Gilbert.

shyeonaaa, 2018-08-06:
thank you! kijoingGil

so you mean that

Should I edit the following that samples24, resolution15?

        <scan>
          <horizontal>
            <samples>24</samples>
            <resolution>15</resolution>
            <min_angle>0.0</min_angle>
            <max_angle>6.28319</max_angle>
          </horizontal>
        </scan>

kijongGil, 2018-08-06:
Just edit sample 24,
```
        <samples>24</samples>
        <resolution>1</resolution>
```
But, If you want to use more or less sample for your learning, you have to change state_size.

robotpilot, 2018-08-12:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

parkjonghyun00, 2019-07-02:
감사합니다

parkjonghyun00, 2019-07-02:
감사합니다 : )