ROBOTIS-GIT/turtlebot3_machine_learning #4: Error: when trying to plot the graph: from PyQt5.QtWidgets import * ImportError: No module named PyQt5.QtWidgets

BowenY, 2018-08-07:
Hi,
I was able to install the machine-learning package, run gazebo and start dqn. However, when I tried to plot the graph using "roslaunch turtlebot3_dqn result_graph.launch", I got errors and will not be able to view the graphs. 
logs:
catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/action_graph", line 21, in <module>
    from PyQt5.QtWidgets import *
ImportError: No module named PyQt5.QtWidgets
/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/result_graph", line 23, in <module>
    from PyQt5.QtGui import *
ImportError: No module named PyQt5.QtGui

I was thinking about the mismatch between pyqt4 and pyqt5 so I installed pyqt5 but there is no change. Also I tried to change import PyQt5.* in action_graph/result_graph files to import PyQt4.* but not effective.

I noticed in the tutorial it install pyqt 4.11.4.

May I know what's going on and how to solve it?

Thanks,
BowenY

kijongGil, 2018-08-08:
Hi @BowenY :)
Maybe this will solve your problem.

```
pip uninstall tensorflow
pip install tensorflow==1.5
pip install pyqtgraph
```
Thanks,
Gilbert.

Vincent7803, 2018-08-08:
Hi Gilbert,
I too had similar issue...I tried the fix that you had recommended..but its doesn't resolve the issue. The same error is reported.

Thanks
Vincent

kijongGil, 2018-08-08:
Hi @Vincent7803 :)
Could you remove anaconda2 folder? and then try to type command i said.

Vincent7803, 2018-08-08:
It works now.
Thanks Gilbert

BowenY, 2018-08-08:
Hi @kijongGil 
It worked perfectly, thanks!

P.S. I also need to re-install the ros dependency packages using "pip install -U rosinstall msgpack empy defusedxml netifaces" after I re-installed anaconda, tensorflow, kersa and pyqtgraph to use gazebo. I dont know if Vincent7803 encountered the same requirement but I would love to note this for anyone else may have this situation.

kijongGil, 2018-08-09:
Thank you for your contribution. 
I will close this issue. If you have a another issue, you can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-09 by kijongGil