ROBOTIS-GIT/turtlebot3_machine_learning #5: Unable to get the Stage 3 (moving obstacle) and Stage 4 (combination obstacle) running as mentioned in the tutorial

Vincent7803, 2018-08-09:
Hi,
The stage 3 (moving obstacle) is a repeat of stage 2, with the obstacles not moving around. The stage 4 demo doesn't work as expected. I don't see any movement of the Turtlebot3 in stage 4. I checked the environment_stage_3.py code and it doesn't seem to have a call to moving obstacle.

Thanks
Vincent

kijongGil, 2018-08-09:
Hi @Vincent7803 :)
Calling moving obstacle is turtlebot3_stage_3.launch in turtlebot3_simulations package.
Could you reinstall turtlebot3_simulations and try again?

Thanks,
Gilbert.

Vincent7803, 2018-08-09:
Hi Gilbert,
Which turtlebot3_simulations package do you need me to run with. I tried the latest one, but no success. I tried with the turtlebot3_simulations-8293XX, but i still face the same issue.

Thanks
Vincent

kijongGil, 2018-08-10:
I'm sorry for your inconvenience.
You're right. it doesn't have a call to moving obstacle. It is my bad.
I updated this package.

please remove and reinstall turtlebot3_machine_learning.

And try follow command
$ roslaunch turtlebot3_gazebo turtlebot3_stage_3.launch
$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_3.launch

Thanks.

Vincent7803, 2018-08-10:
Hi Gilbert,
Stage 4 (combination obstacle) is working well.
Looks like the issue is in Stage 3 only.

Thanks
Vincent

kijongGil, 2018-08-15:
I'm sorry for late reply.
I updated again turtlebot3_simulations.
Please git pull again. Maybe this issue is solve.
Thank.

Vincent7803, 2018-08-16:
Thanks Gilbert. the issue is resolved.

closed 2018-08-16 by robotpilot

kijongGil, 2018-08-21:
You can reopen this issue to show this issue to the users whenever. Thanks.

DafangWang, 2019-01-25:
Hi Gilbert, I have the same problem: 
obstacles of the Stage 3 are not moving and Stage 4(moving obstacles) are also not moving. 
The code seems lack of something. 
Appreciate your help!

DafangWang, 2019-01-28:
I will have to follow the below order if i am to succeed in running the code.
first: roslaunch turtlebot3_dqn turtlebot3_dqn_stage_3.launch
second: roslaunch turtlebot3_gazebo turtlebot3_stage_3.launch
otherwise, the final result is not as expected.

kijongGil, 2019-01-29:
Hi @DafangWang :)
What is your gazebo version?