ROBOTIS-GIT/turtlebot3_machine_learning #7: robot cycling

BowenY, 2018-08-16:
Hi there,

I am not sure if I should put this question here. Please move it if it's not appropriate.

When the robot is being trained on stage_1 and stage_2, should it get to the infinite cycling after training finished? I noticed the robot just keep cycling after couple of episodes and the reward recached to over 1500.

Thanks

kijongGil, 2018-08-17:
Hi @BowenY :)
The agent has to subscribe to the scan topic from gazebo, but the learning is sometimes stopped because gazebo does not publish scan topic. It is a bug.
You can see that the learning has stopped, the action graph is not moving.
If learning stops, try again from the last saved episode.

Thanks,
Gilbert.

BowenY, 2018-08-17:
Hi @kijongGil ,

Thanks for replying. That makes sense to me cause I didn't find any code mentioning cycling around in the code base. 
As to the train from the last saved episode, I just would to double check to make sure I found the place currently:
set the "self.load_model = true" and "self.load_episode = <last saved episode number which can be found in the save_model folder>"

Is that correct?

Thanks!

kijongGil, 2018-08-20:
Yes, right. If you have a 'stage_1_100.h5' file, you can load save model.
```
self.load_model = true
self.load_episode = 100
```

kijongGil, 2018-08-22:
I'm going to close this issue. You can reopen this issue to show this issue to the users whenever.

closed 2018-08-22 by kijongGil

EdgarJP006, 2019-12-13:
your recommendation worked

EdgarJP006, 2019-12-13:
> Hi @kijongGil ,
>
> Thanks for replying. That makes sense to me cause I didn't find any code mentioning cycling around in the code base.
> As to the train from the last saved episode, I just would to [...]

your recommendation worked