ROBOTIS-GIT/turtlebot3_machine_learning #8: 'Question' How to use a saved model in 'Save_model' folder

ChangkiSung, 2018-08-20:
Hi, I just started to learn turtlebot3 machine learning.

Does anyone know how to use saved model files? (stage_1_1500.h5, stage_1_1500.json).

Can I load those files to run a launch file??

Thank you.

kijongGil, 2018-08-20:
Hi @ChangkiSung :)
Maybe this will be help you. https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning/issues/7

Thanks,
Gilbert.

ChangkiSung, 2018-08-20:
Hi @kijongGil 

Thank you for solution. It works.

I have one more quick question about the machine learning for turtlebot3.

I would like to change a real time factor to train much faster on gazebo. Does it matter for the result??

Thanks

kijongGil, 2018-08-21:
You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-21 by kijongGil

parkjonghyun00, 2020-02-17:
How to use a saved model in 'Save_model' folder?