ROBOTIS-GIT/turtlebot3_machine_learning #9: Error: When Checking Input.

ChangkiSung, 2018-09-03:
Hi,

When I run Stage 1, I got this Error "ValueError: Error when checking input: expected dense_1_input to have shape (26,) but got array with shape (362,)"

If I modify state_size parameter to 362, I do not get error. However, I am not sure whether its ok or not.

Also, I have another question about using trained model on next section. (Ex, using trained save_model stage 1 on Stage 2)

I have trained with state_size = 362 in stage 1 and I would like to use trained model to stage 2.

But, I got this error "ValueError: Error when checking input: expected dense_1_input to have shape (362,) but got array with shape (364,)"

I am wondering that is it possible to use trained model on next section?

Thank you.

kijongGil, 2018-09-03:
Hi @ChangkiSung :)
This will be help you. https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning/issues/2

If the input shape of the agent is different, learning can not be done. 
Don't care about any stage, because you can modify input shape by yourself.

Thanks,
Gilbert.

ChangkiSung, 2018-09-04:
Hi @kijongGil Thanks for replying.

It works with modifying state_size parameter.

-------------------------------This is solution in my case ---------------------------------------------------
1. First of all, I modified LDS states to 26. The state_size in node file is 28. Then, we can train stage 1 model.
2. For using trained model of stage 1 to train stage 2, you might have this kind of error "ValueError: Error when checking input: expected dense_1_input to have shape (28,) but got array with shape (30,)"
(Change Save model name stage_1_# to stage_2_#)
3. You can change LDS state parameter to 24. The state_size of stage 2 is 28. 
This worked for me to train stage 2 using save model of stage 2.
Also, it will work for stage 3 and 4.

kijongGil, 2018-09-05:
I am glad that your problem has been resolved. I'm going to close this issue.
If you have any problem, please reopen it.

closed 2018-09-05 by kijongGil