ROBOTIS-GIT/turtlebot3_machine_learning #11: Issue Time is out of dual 32-bit range

dovanhuong, 2018-10-29:
Dear Turtlebot3 team,
I'm having an issue while running dqn machine learning for turtlebot3. after it run for a second it came to had an issue with " time is out of dual 32-bit range" as I copied description in below. Could you please have a look on it and give me your advice how to fix this issue!?
Thank you in advance.
/// Issue description.
Command from DQN training:
kist-cmr@kistcmr-940X3G-930X3G:~$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_3.launch
... logging to /home/kist-cmr/.ros/log/6daabb0e-db21-11e8-a010-052aeba48057/roslaunch-kistcmr-940X3G-930X3G-2351.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kistcmr-940X3G-930X3G:35507/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /stage_number: 3

NODES
  /
    moving_obstacle (turtlebot3_dqn/moving_obstacle)
    turtlebot3_dqn_stage_3 (turtlebot3_dqn/turtlebot3_dqn_stage_3)

ROS_MASTER_URI=http://localhost:11311

process[turtlebot3_dqn_stage_3-1]: started with pid [2368]
process[moving_obstacle-2]: started with pid [2369]
Using TensorFlow backend.
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_1 (Dense)              (None, 64)                1856      
_________________________________________________________________
dense_2 (Dense)              (None, 64)                4160      
_________________________________________________________________
dropout_1 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_3 (Dense)              (None, 5)                 325       
_________________________________________________________________
activation_1 (Activation)    (None, 5)                 0         
=================================================================
Total params: 6,341
Trainable params: 6,341
Non-trainable params: 0
_________________________________________________________________
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_4 (Dense)              (None, 64)                1856      
_________________________________________________________________
dense_5 (Dense)              (None, 64)                4160      
_________________________________________________________________
dropout_2 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_6 (Dense)              (None, 5)                 325       
_________________________________________________________________
activation_2 (Activation)    (None, 5)                 0         
=================================================================
Total params: 6,341
Trainable params: 6,341
Non-trainable params: 0
_________________________________________________________________
[INFO] [1540779779.192717, 0.559000]: Goal position : 0.6, 0.0
[INFO] [1540779804.124338, 15.012000]: Collision!!
[INFO] [1540779804.492929, 15.190000]: Ep: 1 score: -256.66 memory: 65 epsilon: 1.00 time: 0:00:26
[INFO] [1540779811.057318, 3.605000]: Goal!!
Traceback (most recent call last):
[... 64 lines cut ...]
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
[ INFO] [1540779727.445171329, 0.023000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1540779727.489099999, 0.066000000]: Physics dynamic reconfigure ready.
[ INFO] [1540779727.645949424, 0.141000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1540779727.646047582, 0.141000000]: Starting Laser Plugin (ns = /)
[ INFO] [1540779727.648495262, 0.141000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[spawn_urdf-4] process has finished cleanly
log file: /home/kist-cmr/.ros/log/6daabb0e-db21-11e8-a010-052aeba48057/spawn_urdf-4*.log
[ INFO] [1540779727.736957369, 0.141000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1540779727.737100552, 0.141000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1540779727.738137890, 0.141000000]: DiffDrive(ns = //): <tf_prefix> = 
[ WARN] [1540779727.738257190, 0.141000000]: DiffDrive(ns = //): <publishWheelTF> must be either true or false
[ WARN] [1540779727.738284271, 0.141000000]: DiffDrive(ns = //): <publishWheelJointState> must be either true or false
[ WARN] [1540779727.738308214, 0.141000000]: DiffDrive(ns = //): <legacyMode> must be either true or false
[ INFO] [1540779727.739348552, 0.141000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[ INFO] [1540779727.742647201, 0.141000000]: DiffDrive(ns = //): Subscribe to cmd_vel
[ INFO] [1540779727.744078090, 0.141000000]: DiffDrive(ns = //): Advertise odom on odom 
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
Couldn't find a preferred IP via the getifaddrs() call; I'm assuming that your IP address is 127.0.0.1.  This should work for local processes, but will almost certainly not work if you have remote processes.Report to the disc-zmq development team to seek a fix.
[ERROR] [1540779811.093034347, 3.621000000]: Exception thrown while processing service call: Time is out of dual 32-bit range

kijongGil, 2018-10-30:
Hi, @dovanhuong :)
Could you tell me what is gazebo and tensorflow version?

dovanhuong, 2018-10-30:
hello, @kijongGil I'm using Gazebo9 and tensorflow version 1.12.0-rc0. and ROS Kinetic. Could you please help to check this!?

kijongGil, 2018-11-01:
Did you use ubuntu 18.04? Because I use Gazebo7, ubuntu 16.04 and ROS Kinetic.
We are not support ubuntu 18.04 yet. 
I think your problem is Gazebo 9. But, I can not use Gazebo 9 in ubuntu 16.04.
Could you change your environment?

dovanhuong, 2018-11-01:
I think the problem is come from Gazebo 9. I tested again with Gazebo 7 and it worked fine. I'll consider to think about change to my own environment. Thank you so much! ;)

kijongGil, 2018-11-05:
This issue is closing. Please reopen it when you have another issue.
Thanks.

closed 2018-11-05 by kijongGil