ROBOTIS-GIT/turtlebot3_machine_learning #13: Question about the setReward

alexofficial, 2018-11-13:
Hello,

I have a question about the formula used in the setReward:

First in the [video](https://www.youtube.com/watch?v=807_cByUBSI&amp=&amp=&amp=&feature=youtu.be) that you have, the angle is calculated as `angle= pi/2 + action * p/8 + Φ` but in your enviroment stage is calculated as  `angle = -pi / 4 + heading + (pi / 8 * i) + pi / 2`. Why?

Second, there is a formula for calculating tr. I guess this is the (1- caret(angle)) from the [video: 1:42 ](https://www.youtube.com/watch?v=807_cByUBSI&amp=&amp=&amp=&feature=youtu.be)but I do not understand what the tr means and how you come up with formula. Is based on a paper or something?
`tr = 1 - 4 * math.fabs(0.5 - math.modf(0.25 + 0.5 * angle % (2 * math.pi) / math.pi)[0])`

Please, can someone explain to me because it does not make sense why you used this from the code?

kijongGil, 2018-11-20:
Hi @alexofficial :)
I'm sorry for late reply. I had a business trip last week.
Actually, the angle meaning of the video and the code seems to be a bit confusing.
The angle of the video is wrong, `angle = -pi/4 + action * pi/8 + Φ `is right.

kijongGil, 2018-11-27:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-27 by kijongGil

meh369, 2019-12-30:
> Hi @alexofficial :)
> I'm sorry for late reply. I had a business trip last week.
> Actually, the angle meaning of the video and the code seems to be a bit confusing.
> The angle of the video is wrong, `angle = -pi/4 + action * pi/8 + Φ `is right.

Is Φ, goal angle or yaw??

Ac31415, 2022-07-21:
> > Hi @alexofficial :)
> 
> > I'm sorry for late reply. I had a business trip last week.
> 
> > Actually, the angle meaning of the video and the code seems to be a bit confusing.
> 
> > The angle of the video is wrong, `angle = -pi/4 + action * pi/8 + Φ `is right.
> 
> 
> 
> Is Φ,  Heading or yaw??

I have the same question, too!