ROBOTIS-GIT/turtlebot3_machine_learning #14: 'goal' model is missing in gazebo after when turtlebot hits it for the first time

MingyiZhang, 2018-12-06:
Dear turtlebot3 team,

I have a problem when running the simulations. 

After the turtlebot3 gets the goal, the goal doesn't render again in gazebo. I checked the rostopic gazebo/model_states, the goal model is respawned, the robot can hit it during training, but it is not showing in the gazebo... What could be the problem? 

My system is Ubuntu 16.04, ROS Kinetic, Gazebo 7, in VirtualBox, the setup is from a clean image, following the tutorial in the website.

kijongGil, 2019-01-02:
Hi, @MingyiZhang :)
I'm so sorry for late reply.
When you run the gazebo, there is goal model? and after the turtlebot3 gets the goal, the goal model doesn't again respawn?

MingyiZhangQimia, 2019-01-08:
Hi @kijongGil, 

Thank you very much for your reply! 
I am sorry that I didn't update the status. 
I changed the goal model's color to "Gazebo/Red" and the goal model reappears in Gazebo. 
I think it would be great to set the color of the goal model to "Gazebo/Red" in
`turtlebot3_simulations/turtlebot3_gazebo/models/turtlebot3_square/goal_box/model.sdf`

I think you can close this issue :+1:

MingyiZhang, 2019-01-08:
Hi @kijongGil ,

Thank you very much for your reply!
I am sorry that I didn't update the status.
I changed the goal model's color to "Gazebo/Red" and the goal model reappears in Gazebo.
It would be great to set the color of the goal model to "Gazebo/Red" in
`turtlebot3_simulations/turtlebot3_gazebo/models/turtlebot3_square/goal_box/model.sdf`

You can close this issue :+1: :+1: