ROBOTIS-GIT/turtlebot3_machine_learning #15: what the reward function mean？

zengjie617789, 2019-04-09:
is there anyone who can explain what the reward function exactly mean？ thanks in advance~

```
`for i in range(5):
            angle = -pi / 4 + heading + (pi / 8 * i) + pi / 2
            tr = 1 - 4 * math.fabs(0.5 - math.modf(0.25 + 0.5 * angle % (2 * math.pi) / math.pi)[0])
            yaw_reward.append(tr)

        distance_rate = 2 ** (current_distance / self.goal_distance)
        reward = ((round(yaw_reward[action] * 5, 2)) * distance_rate)`
```

RyanJaehyunShim, 2019-04-09:
@zengjie617789 

If you would like to know more details about the code, please read the tutorial 14.2. 
Also, the below video in the link below briefly explains about it (1:40, 1:50) 
(https://www.youtube.com/watch?v=807_cByUBSI)

Regards,
Ryan

zengjie617789, 2019-04-09:
thank you very much， i watched the video, and still confused about the angle theta， because the variable “heading” in that code means the goal angle. from my perspective ，it just like the picture below：
[image]
but in that video ，it shows：
[image]
and that code is：
```
self.position = odom.pose.pose.position
        orientation = odom.pose.pose.orientation
        orientation_list = [orientation.x, orientation.y, orientation.z, orientation.w]
        _, _, yaw = euler_from_quaternion(orientation_list)

        goal_angle = math.atan2(self.goal_y - self.position.y, self.goal_x - self.position.x)

        heading = goal_angle - yaw
        if heading > pi:
            heading -= 2 * pi

        elif heading < -pi:
            heading += 2 * pi

        self.heading = round(heading, 2)
```

RyanJaehyunShim, 2019-04-09:
Would you elaborate your question again? I think you got it correctly.

zengjie617789, 2019-04-09:
@RyanJaehyunShim  I think the angle reward is about calculating the next heading angle after excute an action, but it is a little defferent with the tutorial wrtitten.
here is my understanding below：
[image]
i am sorry about my poor englishi deeply ~~

RyanJaehyunShim, 2019-04-09:
@zengjie617789 

The reward function is defined as in the video and the robot chooses an action with the largest reward. 

The equation you sent is about states of the robot (next state = current state + blahblah).

Regards,
Ryan

zengjie617789, 2019-04-09:
@RyanJaehyunShim  thank you ~ but could you please telll me what this equation mean?
[image]

RyanJaehyunShim, 2019-04-10:
@zengjie617789 

Theta indicates the direction of the robot. On the other hand, there is no specific meaning on the equation about reward function. Reward function can be defined arbitrarily, but as you can see from the equation, R_theta becomes larger when the robot moves towards the goal.

Regards,
Ryan

florianspy, 2019-04-12:
Hello i have a question regarding the reward for distance.
Why is a higher reward given when the robot traveled away from the goal e.g. if current_distance gets bigger a higher reward is given, or is in those cases the anglereward always negative?

zengjie617789, 2019-04-14:
@florianspy  i think the reward for distance will not get larger when the robot away from the goal . As the pictue blow, there is a threshold Dg and i think it is a very small value such as 0.2 or whatever ,because our target is a area .
[image]
i think Dc/Dg is almost higher than 1 so the distant reward is higher than 2 which is the same as the picture shows blow:

[image]
but it is against the second picture~~~

As far as the angle reward , it is not always negative. there is a constant 5 before the equation and the output of theta function is always less than 5.

[image]

kijongGil, 2019-04-15:
@florianspy @zengjie617789
This reward function may not be perfect and may logically be erroneous. When I first set up the reward function, I set it to 1 if I get closer to the goal, or -1 if I get farther. However, I felt the need to subdivide the reward function a bit further because the result was not good. So I set reward function on two conditions.
1. Large reward when the robot's heading is in the goal direction.
2. A small coefficient is obtained when approaching the goal, and a large coefficient is obtained when the distance is away.

I hope you set up a more effective reward function and try to your ML.

zengjie617789, 2019-04-15:
@kijongGil  thank you for your answer. I am still confused that why the distance reward get large when the robot away from the foal （Dc/Dg is larger than 1）.

kijongGil, 2019-04-15:
@zengjie617789 
Because the angler reward is negative when the robot moves away from the goal, it gains a large negative reward follow equation.
total reward = angler reward * distance reward.
I hope that the robot get a big negative reward when moving away from the goal.

florianspy, 2019-04-24:
Thank you for your answers

ROBOTIS-Kayman, 2019-05-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-09 by ROBOTIS-Kayman