ROBOTIS-GIT/turtlebot3_machine_learning #16: Turtlebot fails to learn in the presence of obstacles.

Usama1002, 2019-05-01:
Turtlebot fails to learn in the presence of obstacles. How to get the results shown in the tutorial video?

RyanJaehyunShim, 2019-05-02:
Write more details, or nobody knows what you did and why you didn't get the expected results.

Usama1002, 2019-05-02:
Well, that's the entire point. I just ran the code provided in the repo. I want to know if I am missing out something that you guys additionally used in getting results showed in the tutorial video?

RyanJaehyunShim, 2019-05-02:
Write more details, or nobody knows which code you ran and what error you've got from it. 

Check out the link below about how you should describe the issue you need help with.
(https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning/issues?q=is%3Aissue+is%3Aclosed)

RyanJaehyunShim, 2019-05-16:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if you have any other questions. Thanks.

closed 2019-05-16 by RyanJaehyunShim