ROBOTIS-GIT/turtlebot3_machine_learning #17: How to add a moving obstacle in world file?

zengjie617789, 2019-05-06:
I test the env stage3 successfully when i run turtlebot3_dqn_stage_3 launch file firstly. However when I tried to build a moving obstacle world file in gym_gazebo, it failed.
here is what i tried：
1. gazebo random velocity plugin; http://gazebosim.org/tutorials?tut=plugins_random_velocity&cat=
    but it cannot detect the boundary
2. gazebo actor model; http://gazebosim.org/tutorials?tut=actor&cat=build_robot
    i tried the two way of adding animated model,however it seem like the model doesnot have 
    collision property
i did my best to make the obstacle move and i am exhausted while i googled many times 
anyone who can help me   ? thank you in advance

kijongGil, 2019-05-07:
Hi @zengjie617789 :)
I understand that you spend many time for moving obstacle in gazebo. I also tried several methods and then I found using ModelState topic. I'm not sure It is best way. You can see how to use this topic in our package in this [code.](https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning/blob/master/turtlebot3_dqn/nodes/moving_obstacle)

Thanks,
Gilbert

zengjie617789, 2019-05-07:
@kijongGil  thanks, I tried this method ,but it seems only work in a situation that launch the env file and dqn file separately. However,the gymgazebo algorithm files is combined with the env file. I also tried to run the moving python file alone ,but the obstacle didnot  move at all . Moreover ,when i run the command of "rostopic echo /gazebo/model_states"  ,it cannot output anything.

RyanJaehyunShim, 2019-05-07:
@zengjie617789 

Please ask the question at the link below.
(https://github.com/erlerobot/gym-gazebo)

Please open a new issue If you have any other questions while following the tutorial on our e-manual.
Ryan

closed 2019-05-07 by RyanJaehyunShim