ROBOTIS-GIT/turtlebot3_machine_learning #18: Where can I see the rendering code?

hyunsooda, 2019-05-21:
Hello, I am a student who learn ROS and Reinforcement learning.

I saw the code which is related to RL but I never saw rendering code with opencv(maybe?)

Where can I see these works ? 

Thanks for great tutorial.

kijongGil, 2019-05-22:
Hi @hyunsooda :)
What is rendering code you said? This code don't use openCV.

This code has two parts gazebo environment and RL. I created gazebo environment by https://github.com/erlerobot/gym-gazebo 
And RL code was created by myself.

Thanks,
Gilbert.

hyunsooda, 2019-05-22:
Thanks for response.

Oh... I was confused and didn't know option to render lidar because I'm newbie.
Actually this is provided by default in ROS (xacro file).

I am reviewing your RL code and some points has probability to enhance I think.
I will post new issue when my idea is clear. at that time please comment about my thought

have a nice day Gilbert!

closed 2019-05-22 by hyunsooda