ROBOTIS-GIT/turtlebot3_machine_learning #25: Question: Where is models such as wall, burger, obstacle, box of goal

hyunsooda, 2019-06-14:
Hi :)
I was about to add other goal box but I couldn't find the location where it is.
There is no code related to models in the ```catkin_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models/turtlebot3_burger```

The source code not contains "goal" model not "goal_box". 
```
<?xml version='1.0'?>
<sdf version='1.6'>
  <model name='goal_box'>
    <pose frame=''>0 0 0 0 -0 -1.5708</pose>
    <link name='goal_box'>
      <visual name='goal_box'>
        <pose frame=''>0 0 0.0005 0 -0 0</pose>
        <geometry>
          <box>
            <size>0.5 0.5 0.001</size>
          </box>
        </geometry>
        <material>
          <script>
            <uri>file://media/materials/scripts/gazebo.material</uri>
            <name>Gazebo/Grey</name>
          </script>
          <ambient>1 0 0 1</ambient>
        </material>
      </visual>
      <pose frame=''>0 0 0 0 -0 0</pose>
    </link>
    <static>1</static>
  </model>
</sdf>
```

How could you set the model such as box of goal ? 
Thank you

hyunsooda, 2019-06-15:
I have solved. 
let me know how can I solve it. 

Close.

closed 2019-06-15 by hyunsooda

HamzaNAHID, 2019-08-19:
> I have solved.
> let me know a person who are curious so that I share it.
> 
> Close.

Hello! I am facing the same issue, can you please share how you solved it ?

sharypovandrey, 2020-06-05:
@hyunsooda hi!
Could you let me know how you solved it?

sharypovandrey, 2020-06-05:
I cloned https://github.com/ROBOTIS-GIT/turtlebot3_simulations repo to catkin_ws/src. It has all required files