ROBOTIS-GIT/turtlebot3_machine_learning #26: [turtlebot3_dqn_stage_1-1] process has died

FIsh24, 2019-08-06:
Hi! I'm trying to run this command line: `roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch` but I have this error:

```

ImportError: cannot import name abs
[turtlebot3_dqn_stage_1-1] process has died [pid 13306, exit code 1, cmd /home/fish/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_1 __name:=turtlebot3_dqn_stage_1 __log:=/home/fish/.ros/log/c9d600a2-b828-11e9-9b60-9cb6d08c99cd/turtlebot3_dqn_stage_1-1.log].
log file: /home/fish/.ros/log/c9d600a2-b828-11e9-9b60-9cb6d08c99cd/turtlebot3_dqn_stage_1-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
Please help!

OS: Ubuntu16.04(Xenial)
ROS Version: Kinetic
Python 2.7.16
Turtlebot3 waffle pi
The tutorial I am following: http://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#run-machine-learning

kijongGil, 2019-08-07:
Hi, @FIsh24 :)
I think this problem is installation of tensorflow.
Can you show me the log before the log you posted?

Thanks, 
Gilbert.

FIsh24, 2019-08-07:
I solved the problem with tensorflow. But I still have the same problem:

```
 fish@fish-XPS-13-9370:~/catkin_ws$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_3.launch
... logging to /home/fish/.ros/log/1c01d9d2-b900-11e9-b30d-9cb6d08c99cd/roslaunch-fish-XPS-13-9370-17674.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://fish-XPS-13-9370:44211/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /stage_number: 3

NODES
  /
    moving_obstacle (turtlebot3_dqn/moving_obstacle)
    turtlebot3_dqn_stage_3 (turtlebot3_dqn/turtlebot3_dqn_stage_3)

ROS_MASTER_URI=http://localhost:11311

process[turtlebot3_dqn_stage_3-1]: started with pid [17694]
process[moving_obstacle-2]: started with pid [17695]
Using TensorFlow backend.
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_1 (Dense)              (None, 64)                1856      
_________________________________________________________________
dense_2 (Dense)              (None, 64)                4160      
_________________________________________________________________
dropout_1 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_3 (Dense)              (None, 5)                 325       
_________________________________________________________________
activation_1 (Activation)    (None, 5)                 0         
=================================================================
Total params: 6,341
Trainable params: 6,341
Non-trainable params: 0
_________________________________________________________________
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_4 (Dense)              (None, 64)                1856      
_________________________________________________________________
dense_5 (Dense)              (None, 64)                4160      
_________________________________________________________________
dropout_2 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_6 (Dense)              (None, 5)                 325       
_________________________________________________________________
activation_2 (Activation)    (None, 5)                 0         
=================================================================
Total params: 6,341
Trainable params: 6,341
Non-trainable params: 0
_________________________________________________________________
2019-08-07 12:21:27.893756: I tensorflow/core/platform/cpu_feature_guard.cc:142] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
2019-08-07 12:21:27.919996: I tensorflow/core/platform/profile_utils/cpu_utils.cc:94] CPU Frequency: 1800000000 Hz
2019-08-07 12:21:27.920458: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x55d641c39730 executing computations on platform Host. Devices:
2019-08-07 12:21:27.920501: I tensorflow/compiler/xla/service/service.cc:175]   StreamExecutor device (0): <undefined>, <undefined>
2019-08-07 12:21:28.109384: W tensorflow/compiler/jit/mark_for_compilation_pass.cc:1412] (One-time warning): Not using XLA:CPU for cluster because envvar TF_XLA_FLAGS=--tf_xla_cpu_global_jit was not set.  If you want XLA:CPU, either set that envvar, or use experimental_jit_scope to enable XLA:CPU.  To confirm that XLA is active, pass --vmodule=xla_compilation_cache=1 (as a proper command-line flag, not via TF_XLA_FLAGS) or set the envvar XLA_FLAGS=--xla_hlo_profile.
[INFO] [1565176889.193368, 0.839000]: Goal position : 0.6, 0.0
Traceback (most recent call last):
[... 52 lines cut ...]
[ INFO] [1565176860.950792308]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1565176860.951518782]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1565176861.301894524, 0.024000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1565176861.415386208, 0.133000000]: Physics dynamic reconfigure ready.
[ INFO] [1565176861.506765807, 0.221000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1565176861.935069657, 0.345000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1565176861.939655738, 0.345000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1565176862.236165525, 0.345000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1565176862.236319209, 0.345000000]: Starting Laser Plugin (ns = /)
[ INFO] [1565176862.239164167, 0.345000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1565176862.246360823, 0.345000000]: Starting plugin DiffDrive(ns = //)
[ INFO] [1565176862.247501225, 0.345000000]: DiffDrive(ns = //): <rosDebugLevel> = na
[ INFO] [1565176862.254778420, 0.345000000]: DiffDrive(ns = //): <tf_prefix> = 
[ INFO] [1565176862.257318827, 0.345000000]: DiffDrive(ns = //): Advertise joint_states
[ INFO] [1565176862.259699759, 0.345000000]: DiffDrive(ns = //): Try to subscribe to cmd_vel
[... cut ...]
[ INFO] [1565176862.278264178, 0.345000000]: Physics dynamic reconfigure ready.
[spawn_urdf-3] process has finished cleanly
log file: /home/fish/.ros/log/1c01d9d2-b900-11e9-b30d-9cb6d08c99cd/spawn_urdf-3*.log
Segmentation fault (core dumped)
[gazebo_gui-2] process has died [pid 17173, exit code 139, cmd /opt/ros/kinetic/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/fish/.ros/log/1c01d9d2-b900-11e9-b30d-9cb6d08c99cd/gazebo_gui-2.log].
log file: /home/fish/.ros/log/1c01d9d2-b900-11e9-b30d-9cb6d08c99cd/gazebo_gui-2*.log

```

kijongGil, 2019-08-08:
This problem is set the state of scan data. Please refer to https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning/issues/2

Thanks,
Gilbert.

closed 2019-08-08 by FIsh24