ROBOTIS-GIT/turtlebot3_machine_learning #28: [turtlebot3_dqn_stage_1-1] process has died [pid 1395, exit code -4,

Sebastian5j, 2019-10-14:
I have turtlebot3 and im trying to use CONDA and tensorflow. First i use this command:
roslaunch turtlebot3_gazebo turtlebot3_stage_1.launch
and for some reason i cant see the laser, i know i have to change <xacro:arg name="laser_visual" default="false"/>   and put default="true" but it still doesnt work.
Then when i try roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch
I get the next error :
process[turtlebot3_dqn_stage_1-1]: started with pid [1395]
Using TensorFlow backend.
[turtlebot3_dqn_stage_1-1] process has died [pid 1395, exit code -4, cmd /home/laboratorio/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_1 __name:=turtlebot3_dqn_stage_1 __log:=/home/laboratorio/.ros/log/4baef918-eeba-11e9-91ac-001e4febbb7c/turtlebot3_dqn_stage_1-1.log].
log file: /home/laboratorio/.ros/log/4baef918-eeba-11e9-91ac-001e4febbb7c/turtlebot3_dqn_stage_1-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I have Ubuntu 16.04 xenial
turtlebot3 waffle_pi

rjshim, 2019-10-15:
@Sebastian5j 

1. Check if the file you modified is for the waffle_pi model (there are three, burger, waffle and waffle_pi). Since you are testing waffle_pi, the file below is what you need to modify.
**(turtlebot3/turtlebot3_description/urdf/turtlebot3_waffle_pi.gazebo.xacro)**
2. There are several possible reasons why you are having that issue.
1) You should run gazebo before running that DQN program (and you didn't)
2) test below instead of waffle_pi
export TURTLEBOT3_MODEL=burger
3) The problem is occurring because of Keras related issues. Show us your ~/.bashrc file. 

Regards,
Ryan

Sebastian5j, 2019-10-15:
@rjshim  this is what i have:
in the file turtlebot3_waffle_pi.gazebo.xacro 
<?xml version="1.0"?>
<robot name="turtlebot3_waffle_pi_sim" xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:arg name="laser_visual"  default="true"/>
  <xacro:arg name="camera_visual" default="true"/>
  <xacro:arg name="imu_visual"    default="true"/>

  <gazebo reference="base_link">
    <material>Gazebo/DarkGrey</material>
  </gazebo>

  <gazebo reference="wheel_left_link">
    <mu1>0.1</mu1>
    <mu2>0.1</mu2>
    <kp>500000.0</kp>
    <kd>10.0</kd>
    <minDepth>0.001</minDepth>
    <maxVel>0.1</maxVel>
    <fdir1>1 0 0</fdir1>
    <material>Gazebo/FlatBlack</material>
  </gazebo>

  <gazebo reference="wheel_right_link">
    <mu1>0.1</mu1>
    <mu2>0.1</mu2>
    <kp>500000.0</kp>
    <kd>10.0</kd>
    <minDepth>0.001</minDepth>
    <maxVel>0.1</maxVel>
    <fdir1>1 0 0</fdir1>
    <material>Gazebo/FlatBlack</material>
  </gazebo>

  <gazebo reference="caster_back_right_link">
    <mu1>0.1</mu1>
    <mu2>0.1</mu2>
    <kp>1000000.0</kp>
    <kd>100.0</kd>
    <minDepth>0.001</minDepth>
    <maxVel>1.0</maxVel>
    <material>Gazebo/FlatBlack</material>
  </gazebo>

  <gazebo reference="caster_back_left_link">
    <mu1>0.1</mu1>
    <mu2>0.1</mu2>
    <kp>1000000.0</kp>
    <kd>100.0</kd>
    <minDepth>0.001</minDepth>
    <maxVel>1.0</maxVel>
    <material>Gazebo/FlatBlack</material>
  </gazebo>

  <gazebo reference="imu_link">
    <sensor type="imu" name="imu">
      <always_on>true</always_on>
      <visualize>$(arg imu_visual)</visualize>
    </sensor>
    <material>Gazebo/Grey</material>
  </gazebo>

  <gazebo>
    <plugin name="turtlebot3_waffle_pi_controller" filename="libgazebo_ros_diff_drive.so">
      <commandTopic>cmd_vel</commandTopic>
      <odometryTopic>odom</odometryTopic>
      <odometryFrame>odom</odometryFrame>
      <odometrySource>world</odometrySource>
      <publishOdomTF>true</publishOdomTF>
      <robotBaseFrame>base_footprint</robotBaseFrame>
[... 245 lines cut ...]
 #!! Contents within this block are managed by 'conda init' !!
__conda_setup="$('/home/laboratorio/anaconda2/bin/conda' 'shell.bash' 'hook' 2> /dev/null)"
if [ $? -eq 0 ]; then
    eval "$__conda_setup"
else
    if [ -f "/home/laboratorio/anaconda2/etc/profile.d/conda.sh" ]; then
        . "/home/laboratorio/anaconda2/etc/profile.d/conda.sh"
    else
        export PATH="/home/laboratorio/anaconda2/bin:$PATH"
    fi
fi
unset __conda_setup
 #<<< conda initialize <<<
#COMMENT

........................................
And im running this:
1) roslaunch turtlebot3_gazebo turtlebot3_stage_1.launch
2) roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch
The second one fails

If i try export TURTLEBOT3_MODEL=burger
the first one fails 
I have Keras 2.1.5
and tensorflow 1.2.0

Sebastian5j, 2019-10-15:
@rjshim  i dont what happend D:  but this is my  turtlebot3_waffle_pi.gazebo.xacro
  <xacro:arg name="laser_visual"  default="true"/>
  <xacro:arg name="camera_visual" default="true"/>
  <xacro:arg name="imu_visual"    default="true"/>
I hope this is what you wanted to see

rjshim, 2019-10-16:
@Sebastian5j 

Use burger model. Before running the first line, run $ conda deactivate. Then, open another terminal window and run the second line. Test it and show me the screenshot of your terminal and gazebo windows. 

Regards,
Ryan

Sebastian5j, 2019-10-16:
@rjshim  I did what you said, this happened:
[image]

I also tried to run $conda deactivate in the second windows before $ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch but the same thing happened
[image]

rjshim, 2019-10-16:
@Sebastian5j 

No

1)
$ conda deactivate
$ roslaunch turtlebot3_gazebo turtlebot3_stage_1.launch

2) Open another window and 
$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch

Try it and let me know the result

Sebastian5j, 2019-10-17:
It works now, i found that it was a problem with versions of tensorflow and keras, besides tensorflow wasn't in PYTHONPATH for some reason. Thanks! what you @rjshim  told helped me to find the problem.

closed 2019-10-17 by Sebastian5j