ROBOTIS-GIT/turtlebot3_machine_learning #29: Can't see laser with roslaunch turtlebot3_gazebo turtlebot3_stage_1.launch

Sebastian5j, 2019-10-17:
i'm using waffle_pi so my turtlebot3_waffle_pi.gazebo.xacro file looks like this:

<?xml version="1.0"?>
<robot name="turtlebot3_waffle_pi_sim" xmlns:xacro="http://ros.org/wiki/xacro">
  <xacro:arg name="laser_visual"  default="true"/>
  <xacro:arg name="camera_visual" default="false"/>
  <xacro:arg name="imu_visual"    default="false"/>
.
.
.
 <gazebo reference="base_scan">
    <material>Gazebo/FlatBlack</material>
    <sensor type="ray" name="lds_lfcd_sensor">
      <pose>0 0 0 0 0 0</pose>
      <visualize>$(arg laser_visual)</visualize>
      <update_rate>5</update_rate>
      <ray>
        <scan>
          <horizontal>
            <samples>24</samples>
            <resolution>1</resolution>
            <min_angle>0.0</min_angle>
            <max_angle>6.28319</max_angle>
          </horizontal>
        </scan>
        <range>
          <min>0.120</min>
          <max>3.5</max>
          <resolution>0.015</resolution>
        </range>
        <noise>
          <type>gaussian</type>
          <mean>0.0</mean>
          <stddev>0.01</stddev>
        </noise>
      </ray>
      <plugin name="gazebo_ros_lds_lfcd_controller" filename="libgazebo_ros_laser.so">
        <topicName>scan</topicName>
        <frameName>base_scan</frameName>
      </plugin>
    </sensor>
  </gazebo>

And my .bashrc :
export ROS_MASTER_URI=http://192.168.0.113:11311
export ROS_HOSTNAME=192.168.0.113
export TURTLEBOT3_MODEL=waffle_pi
#export TURTLEBOT3_MODEL=burger

# >>> conda initialize >>>
# !! Contents within this block are managed by 'conda init' !!
__conda_setup="$('/home/laboratorio/anaconda2/bin/conda' 'shell.bash' 'hook' 2> /dev/null)"
if [ $? -eq 0 ]; then
    eval "$__conda_setup"
else
    if [ -f "/home/laboratorio/anaconda2/etc/profile.d/conda.sh" ]; then
        . "/home/laboratorio/anaconda2/etc/profile.d/conda.sh"
    else
        export PATH="/home/laboratorio/anaconda2/bin:$PATH"
    fi
fi
unset __conda_setup
# <<< conda initialize <<<
And this is what appends with $ roslaunch turtlebot3_gazebo turtlebot3_stage_1.launch

[image]

rjshim, 2019-10-18:
@Sebastian5j 

I tested the same thing and it worked as the instruction.

[image]
[image]

Would you just remove all packages and reinstall them and see if it works?
You have to reinstall tb3, tb3_simulations packages.

Regards,
Ryan

Sebastian5j, 2019-10-18:
@rjshim  I did it:
sudo apt-get install ros-kinetic-turtlebot3
sudo apt-get install ros-kinetic-turtlebot3-simulations
Nothing happened, but im surprised because i thougth those packages were already installed

rjshim, 2019-10-21:
@Sebastian5j 
Unfortunately, there is no way I can see what unnecessary modifications you made before, I suggest you reinstall the entire OS, ROS and TB3 packages and test it again. 
Ryan

closed 2019-11-13 by rjshim