ROBOTIS-GIT/turtlebot3_machine_learning #36: Goal is not being achieved for Distant Goal

RiyaPahwa, 2020-01-30:
For nearby goals, turtlebot3 reaches the goal position, but for distant goals, it is not able to reach the goal position.

ROBOTIS-Will, 2020-01-31:
@RiyaPahwa 
Could you be more specific on "not able to reach the goal position for a distant goal"?
It will be much helpful to see the actual issue or any error message on the screen.

RiyaPahwa, 2020-01-31:
There is no error message on screen, but when the goal is far from the turtlebot's position, it does not reach the goal and keeps moving around certain point.

RiyaPahwa, 2020-01-31:
Also, if I keep the goal fixed at some corner of the maze, the turtlebot never reaches the goal and just keeps moving.

ROBOTIS-Will, 2020-02-03:
@RiyaPahwa 
If the navigation is not successful under a specific map, you might want to adjust some navigation parameter so that the robot can navigate in the custom map.
Please see below eManual for tuning navigation parameters.
http://emanual.robotis.com/docs/en/platform/turtlebot3/navigation/#tuning-guide

ROBOTIS-Will, 2020-02-28:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if necessary. Thanks.

closed 2020-02-28 by ROBOTIS-Will