ROBOTIS-GIT/turtlebot3_machine_learning #37: I want to use machine learning

tokarev14, 2020-02-05:
I'm learning machine learning by watching e-manual.

I'd like to do this now on the real turtlebot.

By the way, I passed the command through the remote computer to Turtlebot, and it's not working. (Rroslaunch turtlebot3_dqn turtlebot3_dqn_stage_3.launch)

If you know a solution, I'd appreciate it if you could let me know.

Also, if I could set up for random coordinates, I would appreciate it if you could tell me how.

ROBOTIS-Will, 2020-02-19:
@tokarev14 
The machine learning is designed to be used in Gazebo simulator.
If you want to run it with the actual TurtleBot3, you should modify the source code so that actual robot data can be acquired and delivered to the algorithm.
For example, `scan` is not an actual sensor data, but simulated LDS sensor value.

ROBOTIS-Will, 2020-03-23:
This issue will be closed since there were no actions for a while. Please feel free to reopen this issue when necessary. Thanks.

closed 2020-03-23 by ROBOTIS-Will