ROBOTIS-GIT/turtlebot3_machine_learning #40: Ros2-turtlebot3_machine_learning-Error

arun0303, 2020-03-26:
@rjshim 

Hi Ryan Jaehyun Shim,
**Ubuntu-18.04**
**Ros2 Dashing**
Tried Running Stage 1 (No Obstacle) from the below robotis link
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_machine_learning/
Opened a Terminal:
Tried running the below command
ros2 launch turtlebot3_dqn turtlebot3_dqn_stage1.launch.py
i getting the error attached below.

Thanks 
ARUN
[image]

rjshim, 2020-03-26:
@arun0303 

Would you check if you downloaded the necessary packages and colcon build them properly? If you are not familiar with how to use ROS2 yet, you would better start with ROS2 tutorials first ([Link](https://index.ros.org/doc/ros2/)).

Ryan

arun0303, 2020-03-26:
@rjshim 
Yes i had downloaded the necessary packages and build it as mentioned in the below link and  the build was successful.
http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2_machine_learning/

$ cd ~/robotis_ws/src/
$ git clone -b ros2 https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning.git
$ cd ~/robotis_ws && colcon build --symlink-install

I am not a beginner with ros2.

rjshim, 2020-03-26:
@arun0303 

Would you test below and let me know if it still doesn't work? 
$ ros2 launch turtlebot3_gazebo turtlebot3_dqn_stage1.launch.py
$ ros2 run turtlebot3_dqn dqn_gazebo 1
$ ros2 run turtlebot3_dqn dqn_environment
$ ros2 run turtlebot3_dqn dqn_agent 1

The emanual will be also updated soon. Thank you for the report. 

[image]

Ryan

closed 2020-03-31 by rjshim