ROBOTIS-GIT/turtlebot3_machine_learning #42: Pre-trained model

sharypovandrey, 2020-06-05:
Hi guys,

Does somebody have a pre-trained model?

I'm running stage 1 [step by step](http://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/), but even after 300 epochs, the agent acts pretty randomly.

ashekim, 2020-06-18:
@sharypovandrey 
Hello @sharypovandrey. I'm sorry for the late reply.

Turtlebot3 Machine Learning is using Keras from Tensor Flow.
Please refer to the link below.

[https://www.tensorflow.org/guide/keras/sequential_model](url)
[https://groups.google.com/forum/#!forum/keras-users](url)

Thank you.

sharypovandrey, 2020-06-19:
Thank you @ashekim for your response!
I'm looking for a model that is already trained on stage 4

ashekim, 2020-06-22:
@sharypovandrey 

I would greatly appreciate it.
If you have any question or suggest, please ask to me any time. Thanks :)

sharypovandrey, 2020-06-24:
@ashekim sorry, maybe I didn't explain my question correct. I'm looking for a pre-trained model that works good on stage 4, because now the agent doesn't learn anything. I didn't change the source code. As I saw on the [video](https://www.youtube.com/watch?v=5uIZU8PCHT8&t=100s), agent acts pretty smart. So maybe you guys have this model?

ashekim, 2020-06-30:
@sharypovandrey 

Hi. @sharypovandrey 
I'm sorry for the late reply.
I'll check the issue and get back to you.

Thank you.

chan-1207, 2025-06-10:
Please use the new updated version of the package. If you have any issues, please open an additional issue. We're closing this issue.