ROBOTIS-GIT/turtlebot3_machine_learning #43: How to set other parameters in train

AssassinCrow, 2020-06-12:
Hello, I'm working on this package with ROS kinetic without Anaconda.

I tested this following your directions, but I've been wondering another problem; maybe this would have little thing to do with this page.

I'm planning to modify the number of laser detectors rays from 24 to 340 or 720 for the robustness.

However, I also have a feeling that size of the neural network, whose number of hidden units is 64 on each hidden layer, is not big enough to handle with 340 or 720 of input-size, properly.

So it seems it is needed to be expanded; would it be ok if I increase the number of hidden units by 1200 with 340 or 720 input, for example?

I know this is improper, I'm not sure whether it would go well when I apply DQN into a navigation control.

Are you doing your works with only 24 rays of the laser sensor?

Anyway, I really hope this issue would not bother your works...

Thanks in advance.    :)

HosseinSheikhi, 2020-07-07:
@AssassinCrow Feel free to change the number of hidden layers and neurons in each layer as well. These are the hyperparameters of DQN and you just have to give them a try to find out which one fits better in your case.

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, so we will proceed with closing it.

closed 2025-02-24 by sunghowoo