ROBOTIS-GIT/turtlebot3_machine_learning #44: An objection for `distance_rate` in environment code

AssassinCrow, 2020-06-14:
Hello, I'm working on ROS KInetic with Ubuntu 16.04.

Anyway, according to [here](https://youtu.be/807_cByUBSI), the `distance_rate` which corresponds to reward for distance is bigger than 2 in the case of the current goal-robot distance is shorter than absolute goal-distance.

But in [environment code](https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning/blob/master/turtlebot3_dqn/src/turtlebot3_dqn/environment_stage_4.py), the distance_rate argument is smaller in the upper case; in other words, I have a feeling that it should be modified like,

from `distance_rate = 2 ** (current_distance / self.goal_distance)` 
to `distance_rate = 2 ** (self.goal_distance / current_distance)`

I think the reward would get bigger and bigger when the robot gets farther from the goal point if it moves following the default code.

Would you happen to teach me why you set the `distance_rate` like that?

Thanks in advance. :)

valderrama14, 2020-10-12:
Hi, was this issue ever answered in private?
I'd actually like to know this, because the observation seems valid.

chan-1207, 2025-04-15:
Sorry for the late reply. Our Open Platform Team was newly established in 2025, and we have started maintaining the previously inactive open platform. The Machine Learning repository is also a high-priority task, and we plan to complete its porting to ROS 2 Humble by April. Thank you for your patience

chan-1207, 2025-06-10:
Please use the new updated version of the package. If you have any issues, please open an additional issue. We're closing this issue.