ROBOTIS-GIT/turtlebot3_machine_learning #45: Lidar rays as state input

HosseinSheikhi, 2020-07-06:
I found the state (the input to the deep-Q network) in the ROS2 version weird and confusing. As the documentation says and what is implemented in ROS 1 versions, the state supposed to be a vector by size 26 including 24 rays of lidar. However, in the ROS2 implementation, just min range and arg min of the lidar are extracted and the state is just a vector by size 4! Could you please explain what is going on?

LAXMEKS, 2021-03-06:
yes i too have the same doubt.
how the vector value from lidar is considered in the input state? 
here is my input state value.
[image]

ROBOTIS-Will, 2021-03-08:
Sorry about the discrepancy between the eManual descriptions and the ROS2 source code.
The ROS2 code had been modified to use the distance and angle of the goal position, and those of the nearest obstacle (from the scan data) as an input state o.
I'm currently reviewing the machine learning code as well as the documentations.
Thank you.