ROBOTIS-GIT/turtlebot3_machine_learning #51: Error launching DQN_stage3 and DQN_stage4

aveen-d, 2020-11-16:
Hi I followed all the steps mentioned in the turtlebot3 manual and I get this error whenever I try to launch dqn_stage3 and dqn_stage4. I am noob at ro2, please help me fix this.

System details:
Ubuntu 18.04
ROS2 dashing
[image]

linzhuobin-fighter, 2020-12-22:
hi, I met with this problem too.
after I :

```
export GAZEBO_PLUGIN_PATH=$HOME/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models/turtlebot3_dqn_world/obstacle_plugin/build:$GAZEBO_PLUGIN_PATH
```

problem solved. It looks like the path to the gazebo plugin(libobstacle1.so and libobstacle2.so) is missing and  gazebo can't load them which caused this problem.

chan-1207, 2025-04-15:
Hello,
I apologize for the delay in responding.

Please add the plugin pass as per the answer above.

sunghowoo, 2025-06-10:
This issue has been inactive for a long time, so I will close it.
Please feel free to reopen it if the problem persists.

closed 2025-06-10 by sunghowoo