ROBOTIS-GIT/turtlebot3_machine_learning #52: Error on respawnGoal

LAXMEKS, 2021-01-19:
Traceback (most recent call last):
  File "/home/user/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_2", line 30, in <module>
    from src.turtlebot3_dqn.environment_stage_2 import Env
  File "/home/user/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/src/turtlebot3_dqn/environment_stage_2.py", line 29, in <module>
    from respawnGoal import Respawn
ModuleNotFoundError: No module named 'respawnGoal'
[turtlebot3_dqn_stage_2-1] process has died [pid 9975, exit code 1, cmd /home/user/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_2 __name:=turtlebot3_dqn_stage_2 __log:=/home/user/.ros/log/26eceb94-5a7a-11eb-8e58-34f64bbf17f1/turtlebot3_dqn_stage_2-1.log].
log file: /home/user/.ros/log/26eceb94-5a7a-11eb-8e58-34f64bbf17f1/turtlebot3_dqn_stage_2-1*.log

how can i call the respawn function fron respawnGoal file?

ROBOTIS-Will, 2021-02-24:
Hi @LAXMEKS 
Sorry about the delayed response.
This package has not been maintained for a while and may need some update to support the latest version of dependent packages.
Please allow me a few weeks to review and update the package.

LAXMEKS, 2021-02-24:
Okay, thanks for informing me.

LAXMEKS, 2021-03-06:
does the packages are ready to use now?

version:
ubuntu 18.04
ROS melodic

ROBOTIS-Will, 2021-03-08:
Thanks for the ping.
I'm currently reviewing the package and the eManual documentation, and hopefully they will be updated by the end of this week.
Is Melodic the version you are interested in?
Or are you also interested in other latest versions such as ROS1 Noetic or ROS2 Foxy?

dhairya02, 2022-04-21:
Hi, 
I am getting the same error and I am interested in ROS Noetic for Ubuntu 20. Could you please help?

Heritagecll, 2024-05-20:
Hello, I have also encountered the same problem as you. Have you resolved it?