ROBOTIS-GIT/turtlebot3_machine_learning #53: Machine learning software setup error

stevensu1838, 2021-02-20:
Hi team,

I followed your instructions on your e manual to install tensorflow and keras on my ROS kinetic PC. However
[e manual ](https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning)
[TurtleBot3 49 Machine Learning tutorial 1](https://www.youtube.com/watch?v=s0qgunKt654)

However, when I run import tensorflow as tf I get the following errors. I am setting up this course for my students. Could you please update your e manual and make it reliable? Thanks a lot. 

`mario@mario:~/Downloads$ python
Python 2.7.15 |Anaconda, Inc.| (default, May  1 2018, 23:32:55) 
[GCC 7.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/__init__.py", line 24, in <module>
    from tensorflow.python import pywrap_tensorflow  # pylint: disable=unused-import
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/python/__init__.py", line 47, in <module>
    import numpy as np
  File "/home/mario/anaconda2/lib/python2.7/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/home/mario/anaconda2/lib/python2.7/site-packages/numpy/core/__init__.py", line 91, in <module>
    raise ImportError(msg.format(path))
ImportError: Something is wrong with the numpy installation. While importing we detected an older version of numpy in ['/home/mario/anaconda2/lib/python2.7/site-packages/numpy']. One method of fixing this is to repeatedly uninstall numpy until none is found, then reinstall this version.
>>> import tensorflow as tf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/__init__.py", line 24, in <module>
    from tensorflow.python import pywrap_tensorflow  # pylint: disable=unused-import
  File "/home/mario/anaconda2/lib/python2.7/site-packages/tensorflow/python/__init__.py", line 47, in <module>
    import numpy as np
  File "/home/mario/anaconda2/lib/python2.7/site-packages/numpy/__init__.py", line 140, in <module>
    from . import _distributor_init
ImportError: cannot import name _distributor_init
>>> 
`

The last question is :
Why is it that u created a virtual environment called tensorflow but never activated it? Does it mean that ur tensorflow is installed in the root/base environment? In this case whats the use of creating a virtual env?

This question is asked on your youtube tutorial as well not answered yet. Cheers

ROBOTIS-Will, 2021-02-22:
Hi @stevensu1838 
Thank you for reporting the issue.
I'll review this on my Kinetic machine and get back to you.
Thank you for your patience.

stevensu1838, 2021-02-22:
Hi Will,

Thanks a lot for your great support. I am using exactly Kinetic machine. May I please provide you more details on my errors?

First, I didn't understand why we created a virtual environment called tensorflow but never activated it. So I simply skipped installing anaconda and creating the virtual environment called tensorflow. I installed tensowflow and keras directly on the PC environment.

 
Then in terminal 1:export TURTLEBOT3_MODEL=burger

$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_2.launch

In terminal 2: roslaunch turtlebot3_dqn result_graph.launch

in terminal 3:export TURTLEBOT3_MODEL=burger

$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_2.launch

Then in terminal 3 I got the following errors 

`user@mech1552:~/catkin_ws/src$ roslaunch turtlebot3_dqn turtlebot3_dqn_stage_2.launch
... logging to /home/user/.ros/log/4e84af5c-74ba-11eb-9885-6451065039f7/roslaunch-mech1552-9275.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/user/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://192.168.0.100:33492/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
 * /stage_number: 2

NODES
  /
    turtlebot3_dqn_stage_2 (turtlebot3_dqn/turtlebot3_dqn_stage_2)

ROS_MASTER_URI=http://192.168.0.100:11311

process[turtlebot3_dqn_stage_2-1]: started with pid [9294]
Using TensorFlow backend.
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_1 (Dense)              (None, 64)                1856      
_________________________________________________________________
dense_2 (Dense)              (None, 64)                4160      
_________________________________________________________________
dropout_1 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_3 (Dense)              (None, 5)                 325       
_________________________________________________________________
activation_1 (Activation)    (None, 5)                 0         
=================================================================
Total params: 6,341
Trainable params: 6,341
Non-trainable params: 0
_________________________________________________________________
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
dense_4 (Dense)              (None, 64)                1856      
_________________________________________________________________
dense_5 (Dense)              (None, 64)                4160      
_________________________________________________________________
dropout_2 (Dropout)          (None, 64)                0         
_________________________________________________________________
dense_6 (Dense)              (None, 5)                 325       
_________________________________________________________________
activation_2 (Activation)    (None, 5)                 0         
=================================================================
Total params: 6,341
Trainable params: 6,341
Non-trainable params: 0
_________________________________________________________________
2021-02-22 16:02:48.664099: I tensorflow/core/platform/cpu_feature_guard.cc:140] Your CPU supports instructions that this TensorFlow binary was not compiled to use: AVX2 FMA
[INFO] [1613962969.631518, 0.832000]: Goal position : 0.6, 0.0
Traceback (most recent call last):
  File "/home/user/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_2", line 174, in <module>
    agent.trainModel()
  File "/home/user/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_2", line 121, in trainModel
    q_value = self.model.predict(states.reshape(1, len(states)))
  File "/home/user/.local/lib/python2.7/site-packages/keras/models.py", line 1025, in predict
    steps=steps)
  File "/home/user/.local/lib/python2.7/site-packages/keras/engine/training.py", line 1817, in predict
    check_batch_axis=False)
  File "/home/user/.local/lib/python2.7/site-packages/keras/engine/training.py", line 123, in _standardize_input_data
    str(data_shape))
ValueError: Error when checking : expected dense_1_input to have shape (28,) but got array with shape (364,)
[turtlebot3_dqn_stage_2-1] process has died [pid 9294, exit code 1, cmd /home/user/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_2 __name:=turtlebot3_dqn_stage_2 __log:=/home/user/.ros/log/4e84af5c-74ba-11eb-9885-6451065039f7/turtlebot3_dqn_stage_2-1.log].
log file: /home/user/.ros/log/4e84af5c-74ba-11eb-9885-6451065039f7/turtlebot3_dqn_stage_2-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
user@mech1552:~/catkin_ws/src$ 
`
and the graph is like below. NO lines appear in the plots 
[image]

Thanks a lot

stevensu1838, 2021-02-24:
Hi Will,
May I please ask if you figured this out? I've been trying hard. But I am not good enough to debug this. Thank you and your team very much

ROBOTIS-Will, 2021-02-24:
Hi @stevensu1838 
I've been testing the package in different ROS versions, and it looks like softwares listed in the eManual are outdated and does not supported very well.
I'm working on fixing the issue along with turtlebot3_simulations package, and please allow me a few weeks.
Thank you for your patience.

ROBOTIS-Will, 2021-03-18:
@stevensu1838 
Based on my trials with Kinetic and Melodic, there were some discrepancies in installing the latest Anaconda as the default Python version for those OS were Python2.
In my case, using a virtual environment made it hard to run ROS due to some ros pkg issue.

Regarding the error you are getting from launching `roslaunch turtlebot3_dqn turtlebot3_dqn_stage_2.launch` can be resolved by modifying the number of `scan` samples from `turtlebot3/turtlebot3_description/urdf/turtlebot3_burger.gazebo.xacro` file as described in the eManual below.
Since states receives the scan data, discrepancy in its data length causes the problem when running the node.
https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#set-state

I'm sorry about small delays, but I'll try to complete it within a few days.
Thanks!

ROBOTIS-Will, 2021-03-24:
@stevensu1838 
The eManual instruction for Machine Learning had been updated for ROS1 Kinetic & Melodic.
https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#machine-learning
Please let us know if there's any issue with following the instructions.
Thanks!

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, so I will proceed with closing it.

closed 2025-02-24 by sunghowoo