ROBOTIS-GIT/turtlebot3_machine_learning #56: Robot stuck/ slow when training

Qtsho, 2021-04-23:
Hi, 
Thanks so much for the repos,

I have a problem that when the memory is >64, then the script start to train the NN model. This makes the robot use the last cmd_vel is was told. The cmd_vel only be updated when the NN is trained after one step.

How can I parallel this work? 

I am using stage 4 and  GPU for running the script.

Thanks

ROBOTIS-Will, 2021-04-26:
Hi @Qtsho 
Thank you for your inquiry.
Could you give us more information about your environment and the version of ROS you are running?

Qtsho, 2021-04-26:
> Hi @Qtsho
> Thank you for your inquiry.
> Could you give us more information about your environment and the version of ROS you are running?

Qtsho, 2021-04-26:
Hi,

I am using ROS1 melodic on Ubuntu 18.04.

tensorflow 2

I put some debuging line in the code and see it stucks for a long time between the for loop looping through the batch

` for i in range(self.batch_size):
            states = mini_batch[i][0]
            actions = mini_batch[i][1]
            rewards = mini_batch[i][2]
            next_states = mini_batch[i][3]
            dones = mini_batch[i][4]

            q_value = self.model.predict(states.reshape(1, len(states)))
            self.q_value = q_value

            if target:
                next_target = self.target_model.predict(next_states.reshape(1, len(next_states)))

            else:
                next_target = self.model.predict(next_states.reshape(1, len(next_states)))

            next_q_value = self.getQvalue(rewards, next_target, dones)

            X_batch = np.append(X_batch, np.array([states.copy()]), axis=0)
            Y_sample = q_value.copy()

            Y_sample[0][actions] = next_q_value
            Y_batch = np.append(Y_batch, np.array([Y_sample[0]]), axis=0)

            if dones:
                X_batch = np.append(X_batch, np.array([next_states.copy()]), axis=0)
                Y_batch = np.append(Y_batch, np.array([[rewards] * self.action_size]), axis=0)
` 

This causes the robot stuck for a while a use the last vel_cmd/. As well as slow down the process a lot

matheuslara, 2023-02-23:
Hello.. I am having the same problem with ROS2 Dashing or Foxy. Did you find any solution for this problem?

Zhefan-Xu, 2024-01-30:
I am watching the memroy usage and it is obvious the memory usage is keeping increasing.

sunghowoo, 2025-06-10:
This issue has been inactive for a long time, so I will close it.
Please feel free to reopen it if the problem persists.

closed 2025-06-10 by sunghowoo