ROBOTIS-GIT/turtlebot3_machine_learning #60: turtlebot3_machine_learning for Noetic

nullbyte91, 2021-08-30:
Hi Team,

I'm able to train the dqn algorithm with stage 1 env on Ubuntu 20.04 with few changes. Is there any reason why we haven't release this package for Noetic?

Please let me know if I can contribute those changes.

-
Jegathesan S

ROBOTIS-Will, 2021-09-06:
Hi @nullbyte91 

Thank you for your suggestions.
The machine learning is scheduled after other imminent projects, but your contribution is much appreciated! :+1: 
Based on your PR, I'll schedule an update of the eManual content.
Thanks!

supat-roong, 2022-12-29:
Hi @nullbyte91,

Can you share what did you change, I am also interested in using this package in Ros Noetic with Ubuntu 20.04.

chan-1207, 2025-04-15:
Sorry for the late reply. Our Open Platform Team was newly established in 2025, and we have started maintaining the previously inactive open platform. The Machine Learning repository is also a high-priority task, and we plan to complete its porting to ROS 2 Humble by April. Thank you for your patience

closed 2025-06-10 by sunghowoo