ROBOTIS-GIT/turtlebot3_machine_learning #61: ImportError: libcublas.so.9.0: cannot open shared object file: No such file.

Crawford-fang, 2021-11-12:
ImportError: libcublas.so.9.0: cannot open shared object file: No such file or directory

Failed to load the native TensorFlow runtime.

See https://www.tensorflow.org/install/install_sources#common_installation_problems

for some common reasons and solutions.  Include the entire stack trace
above this error message when asking for help.
[turtlebot3_dqn_stage_4-2] process has died [pid 7852, exit code 1, cmd /home/syzx/catkin_ws/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_4 __name:=turtlebot3_dqn_stage_4 __log:=/home/syzx/.ros/log/de7523e8-4355-11ec-9868-a4bb6d510def/turtlebot3_dqn_stage_4-2.log].
log file: /home/syzx/.ros/log/de7523e8-4355-11ec-9868-a4bb6d510def/turtlebot3_dqn_stage_4-2*.log

Crawford-fang, 2021-11-12:
I use tensorflow-cpu.why can it have that problem.

ROBOTIS-Will, 2021-11-15:
Hi @Crawford-fang 

I'm sorry but we do not provide support for the error with the tensorflow installation.
Have you followed the instruction from the eManual below?
If you have, make sure that you have selected the correct ROS version at the top of the page.
https://emanual.robotis.com/docs/en/platform/turtlebot3/machine_learning/#software-setup

sunghowoo, 2025-02-24:
This issue has been inactive for a long time, so we will proceed with closing it.

closed 2025-02-24 by sunghowoo