ROBOTIS-GIT/turtlebot3_machine_learning #65: ImportError: cannot import name 'Respawn' from 'respawnGoal'

mabijicom1, 2023-10-06:
Hi I am trying to run Machine Learning demos for ros noetic on Ubuntu 20.04 and I am getting this error. When I run this command: roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch, the error burst.This function package uses python2.7, which is inconsistent with the python3 used by noetic. I think this may be the cause of the error, but I don’t know how to solve the error. Please help.
This is the function package I run：
https://github.com/ROBOTIS-GIT/turtlebot3_machine_learning.git

This is the error:
Traceback (most recent call last):
  File "/home/mds/catkin_turtlebot3/devel/lib/turtlebot3_dqn/turtlebot3_dqn_stage_1", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/mds/catkin_turtlebot3/src/turtlebot3_machine_learning/turtlebot3_dqn/nodes/turtlebot3_dqn_stage_1", line 30, in <module>
    from src.turtlebot3_dqn.environment_stage_1 import Env
  File "/home/mds/catkin_turtlebot3/src/turtlebot3_machine_learning/turtlebot3_dqn/src/turtlebot3_dqn/environment_stage_1.py", line 29, in <module>
    from respawnGoal import Respawn
ImportError: cannot import name 'Respawn' from 'respawnGoal' (/home/mds/catkin_turtlebot3/devel/lib/turtlebot3_dqn/respawnGoal.py)
[turtlebot3_dqn_stage_1-1] process has died [pid 23312, exit code 1, cmd /home/mds/catkin_turtlebot3/devel/lib/turtlebot3_dqn/turtlebot3_dqn_stage_1 __name:=turtlebot3_dqn_stage_1 __log:=/home/mds/.ros/log/d236defe-644c-11ee-af09-69f22d66c8d7/turtlebot3_dqn_stage_1-1.log].
log file: /home/mds/.ros/log/d236defe-644c-11ee-af09-69f22d66c8d7/turtlebot3_dqn_stage_1-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

LonFUNZ, 2023-12-20:
Modify the import statement in the environment_stage_1.py file from 'from respawnGoal import Respawn' to 'from src.turtlebot3_dqn.respawnGoal import Respawn .
I using this on Noetic,and it fix the problem,but perhaps this change might have led to another error as well.

mabijicom1, 2023-12-21:
> Modify the import statement in the environment_stage_1.py file from 'from respawnGoal import Respawn' to 'from src.turtlebot3_dqn.respawnGoal import Respawn . I using this on Noetic,and it fix the problem,but perhaps this change might have led to another error as well.

Thanks! it works.

closed 2023-12-21 by mabijicom1