ROBOTIS-GIT/turtlebot3_machine_learning #66: Noetic run turtlebot3_dqn__stage_1.launch  some Issues

LonFUNZ, 2023-12-20:
I can run roslaunch turtlebot3_dqn turtlebot3_dqn_stage_1.launch on Noetic and it executes successfully a few times, but afterwards, I encounter an issue with too many steps, and the turtlebot3 just spins in place.
[image]
I am using Ubuntu 20.04, ROS Noetic, and Anaconda3 (the current version available on the official website),there is also the latest version of tensorflow

Pratik-554, 2024-09-03:
I am facing similar issue. Have you found solution for this?

chan-1207, 2025-04-15:
Sorry for the late reply. Our Open Platform Team was newly established in 2025, and we have started maintaining the previously inactive open platform. The Machine Learning repository is also a high-priority task, and we plan to complete its porting to ROS 2 Humble by April. Thank you for your patience

chan-1207, 2025-06-10:
Please use the new updated version of the package. If you have any issues, please open an additional issue. We're closing this issue.

sunghowoo, 2025-06-10:
This issue has been inactive for a long time, so I will close it.
Please feel free to reopen it if the problem persists.

closed 2025-06-10 by sunghowoo