ROBOTIS-GIT/turtlebot3_machine_learning #70: How to test the saved model ( .h5 and the .jason files )

youssef66677, 2025-02-08:
Hi, I am wondering how to use the saved files produced by the learning (the .h5 and the .jason)  to test the saved learned model if iw ant to make a comparison with an algorithm as the A*.  If anyone could consider this question I will be appreciate it.

Thank you in advance.

sunghowoo, 2025-02-24:
@youssef66677 Sorry for the late reply. Our Open Platform Team was newly established in 2025, and we have started maintaining the previously inactive open platform. The Machine Learning repository is also a high-priority task, and we plan to complete its porting to ROS 2 Humble by April. Thank you for your patience

youssef66677, 2025-03-01:
@sunghowoo If you reached out to any methodology on how to test the navigation of the robot in Gazebo through the generated .h5 file, please contact me as soon as possible. Thank you

chan-1207, 2025-04-15:
The updated package will add the ability to test the trained model.

chan-1207, 2025-06-10:
You can test the newly updated package [turtlebot3_dqn] with dqn_test.py.